cjbaar/prusa-mmu-12x #1: design question

kml-pt, 2021-01-30:
Impressive project! What CAD platform did you use for the design? Would you be willing to share the CAD files? I would like to put different sized fittings on the inlet.

cjbaar, 2021-02-17:
Sorry, not currently prepared to share the source files. There are options for m6 and m10 fittings, and someone else did a mix for a push fitting like the original Prusa one uses. What fitting are you looking to use?

kml-pt, 2021-02-17:
I like to tap an unthreaded 4.2mm hole to use with m5 fittings. But I went ahead with your m10 design and liked it more than I thought I would.

cjbaar, 2021-02-17:
The m10 is definitely my favorite, as it lets me get the PTFE all the way up to the exit point, so IMO less risk of catching on the fitting itself. Another user made a re-mix of selector for 12x, but allows the use of the heat-set insert like Prusa used. See if this is what you are looking for. I haven't added it the repo yet.
[HeatSet-mmu12x-selector-v1.1.stl.zip](https://github.com/cjbaar/prusa-mmu-12x/files/5997583/HeatSet-mmu12x-selector-v1.1.stl.zip)

closed 2022-12-31 by cjbaar

3d-gussner, 2024-10-24:
@cjbaar Do you plan to provide the cad files at some point? If not a problem, just asking before trying to convert the STL files to STEP or cad files manually.

cjbaar, 2024-10-24:
Sorry, not at this time. Is there something specific you are looking to modify?

3d-gussner, 2024-10-25:
> Sorry, not at this time. Is there something specific you are looking to modify?

No problem. I was looking at the STL files and the config changes you have made.
If I am not mistaken the idler bearings are 25° of each other but in the config.h you have 25.4° See https://github.com/cjbaar/Prusa-Firmware-MMU-12x/blob/main/src/config/config.h#L213

Maybe I didn't convert the STL correctly to a STEP/ Fusion360 and am there off by 0.4°

cjbaar, 2024-10-25:
They are 25.2° by design, but I found that 25.4° worked better in the code for proper alignment. Something about the motor steps, perhaps? It's a very small adjustment.

3d-gussner, 2024-10-28:
> They are 25.2° by design, but I found that 25.4° worked better in the code for proper alignment. Something about the motor steps, perhaps? It's a very small adjustment.

Calculating the degrees to actual steps is very hard. The NEMA has 200 Steps per rotation and so 1.8° per full step. With microsteps you get it better. The 32u4 has very limited resources and calculating floats isn't easy here.
Actually your 25.2° are "perfect" full steps (25.2/1.8=14) but you never know what the home position is and so the stepper is probably all the time in some microsteps.

3D printed parts + bearings + tolerances add up quickly and often the "perfect" 25.2° in CAD are in real world 25.4°