cjbaar/prusa-mmu-12x #2: Nice work, impressive

TheZeroBeast, 2021-02-03:
Hey mate,
Nice work, looks like a nice solution.

I'd like to discuss any issues you've faced in your journey with this mod.

Would be cool incorporating some of the software mods I've done to make the MMU2S more reliable.
https://github.com/TheZeroBeast/TZB-MMU2S-Firmware

cjbaar, 2021-02-17:
Thanks for sharing. My fork of the MMU code is so far off the original now, I doubt I will ever try to merge it back. I can't even get them to respond to the relatively small changes needed in the main printer firmware to support change codes over T4. I ended up taking out the eeprom stuff in my version. At the time I didn't feel like digging into the code there enough to understand it, and didn't want to mess with calibrating 12 filaments. Also needed every bit of program space to get the display running. Might look at it again later. In my case, I just adjust it by setting the value in the firmware that works for my PTFE length.

TBH, it never made much sense to me why I would need to calibrate every filament separately. They all through the same sensor, and the size of the selector and length of PTFE don't change.

TheZeroBeast, 2021-02-17:
Yeah, they don't seam to take on much of the assistance provided to them. I tried on a number of fronts for my code and hardware changes to MK3S parts I thought beneficial to the community and zero contact back.

My code only uses the one calibration length and then starts a synced sense portion that runs the MMU pulley synced with printer extruder to grab filament.

cjbaar, 2021-02-17:
Cool; I'll have to take a closer look at it. I pretty much re-wrote all of the load/unload and speed calculations from scratch. They had a lot off weird hard-coded numbers and break points in there, imo. One thing I did was slow down a lot more at the end, to try and match the speed I saw in the main firmware code. I did feel in the original MMU, final feed speed was still a fair amount faster than the extruder gears were running in "wait" mode, which created a higher risk of stripping, even with perfect length calibration. Sounds like that's the same idea you are working with.

flinke-flasche, 2021-04-15:
Really nice work. Thumb up.

I'm thinking about to print all parts and add them to my second Prusa MK3s. 
On the first i have the original MMU2s.

Someday i have to use 12 colors in one print 😄

What are the differnces in both MMU2s firmware files you uploaded?

Why you don’t upload the changed Prusa Firmware for more then T4?

Thank you
Tobias

cjbaar, 2021-04-15:
For the MMU, one has the display integrated (SSD1306) one does not. For the one without display, the buttons use the original Prusa operating code.
I have uploaded a compiled .hex of the printer firmware with support for 12 filaments, but again, I have not tested uploading this. I hav only flashed using the Arduino IDE.

WarrenSchultz, 2021-04-15:
I've ordered the parts to give this a whirl (with display). 
So, as I understand it, unless Prusa implements a major breaking change to the MMU2/MK3 firmware combination, I can use any stock or custom compiled firmware with your MMU2 firmware, as long as I make the tool changes necessary for the Mk3?

cjbaar, 2021-04-15:
@WarrenSchultz Yes; the printer firmware has relatively little changes. The biggest of which is that the original has a hard-coded cap of reading anything above T4. So if you want to patch another version, yes that should work. I was hoping someone would have looked at my PR by now, since it doesn't affect much.

The MMU firmware, however, was heavily modified, so I don't have a PR for that; you would need to use or branch from my fork.

WarrenSchultz, 2021-04-15:
I went and found the PR, looks like there are a couple conflicts now vs. when you did the initial PR, since they did a huge refactor since then. Maybe taking care of those will allow it to sail through? (Especially if those who are following this project boost it.)

I've looked at the MMU2 code before, and I don't have any interest in second-guessing your work on it :D

cjbaar, 2021-04-15:
I can see that conflicts exist; can't see easily what they are. Looks some changes to menus? Maybe they moved to loops like I did, instead of hard-coding 1-5. Realistically, you could make the changes to Marlin code to accept T>5 and that would be enough. The menu changes are only necessary if you ever use the "Load filment X" which I never do. When not printing multi-color, I usually just disconnected my MMU

flinke-flasche, 2021-04-16:
> I've ordered the parts to give this a whirl (with display).
>
> So, as I understand it, unless Prusa implements a major breaking change to the MMU2/MK3 firmware combination, I can use any stock or [...]

Where do you ordered the parts?
Could you order everything in one shop?

zonkie, 2021-04-16:
Guys this is so awesome! I read about this and i follow your discussion actively. I'm not yet sure, if i want to rip my mmu2s apart, but if you get this running, i think i might give it a shot. Maybe we can try to implement the mk3 firmware changes in a configurable way, so Pruša might accept the pull request (unlikely, but worth a shot ;).

cjbaar, 2021-04-16:
@flinke-flasche Pretty sure I just got all the "additional" parts on Amazon, but has to use most of the parts from the stock Prusa unit. If you try to source everything from scratch, I could not find anywhere to buy things like the Prusa custom motors (w/ 100m shaft or screw). Also, I have seen knockoff control boards, but someone else I know that bought one said it didn't work right with the firmware. So, basically, I am not aware my 12x unit can be built unless you already bought the Prusa one. Happy to hear if anyone else has success otherwise.

cjbaar, 2021-04-16:
Just curious... where are people finding or hearing about this build? Never really expected so much interest in it :)

WarrenSchultz, 2021-04-16:
> Just curious... where are people finding or hearing about this build? Never really expected so much interest in it :)

I stumbled onto it a while back, I can't remember where, and forgot about it. Then I saw that TZB had commented on it (whose firmware for the MMU2/Mk3 I use/follow on github), and I remembered I wanted to investigate further. I've been perpetually annoyed by the MMU not having its own display for error codes, and that alone was enough for me to be interested.

zonkie, 2021-04-16:
I cant remember where, i was looking for a color change within a toolhead in the mmu2 and found this... I believe it was mentioned in the Pruša forums.

flinke-flasche, 2021-04-16:
I think someone posted it in a facbook group because of the  presentation of the palette 3.

cjbaar, 2021-04-16:
@WarrenSchultz So, I never tested this because I ripped my MMU apart, but if you just want the screen and error-handling I wrote, you *should* be able to use my MMU firmware with the stock Prusa model. I tried to write everything with parameters, so the motor directions, steps, etc. all are configurable. If you want to try it out, you can compile with the "prusa-mmu2s-display.h" config file

flinke-flasche, 2021-04-19:
> @flinke-flasche Pretty sure I just got all the "additional" parts on Amazon, but has to use most of the parts from the stock Prusa unit. If you try to source everything from scratch, I could not [...]

I would like to test your build with a MMU2 from AliExpress, especially this one 

https://a.aliexpress.com/_veXkkR

cjbaar, 2021-04-19:
@flinke-flasche That's interesting. I didn't full knock-offs like this existed, though I guess I shouldn't be surprised. I am interested in knowing how that turns out, especially with regard to the MCU board.

zonkie, 2021-05-23:
Alright, as this looks like it's working, i started printing the parts. I'm still hesitant about disassembling the original one, but when i oder a knockoff, im having the original one flying around without any use... So I'm probably going that route. Still have some time to decide, the prints will there their time ;) 

Is one 1kh roll of PETG enough?

flinke-flasche, 2021-05-23:
@zonkie 
1kg ist enough.

i thought i will buy a copy from Aliexpress but then i disassembled the original one. 
If you have a lot of screws and nuts you don't have to disassembling the complete MMU2. The only thing is that you have to cut the brass tubes. But with 5 mm less you can use it in your original as well.

cjbaar, 2021-05-26:
@zonkie According to Prusa Slic3r, all the printed parts should be somewhere around 250g total.

Mirarkitty, 2021-12-28:
Very nice work. I'd like to build one of those... how did the aliexpress build work out? 

As for calibrating every filament, it's to calibrate every idler. From when it was super sensitive on the millimeter that it would hit the idler of the extruder, and MMU idler diameter differences were enough to make it unstable. No longer needed - now triggering on the sensor of the extruder instead...

Johnday29, 2022-04-01:
Is this still an active project?  Have you updated it to work with any of the current firmware or slicer?

cjbaar, 2022-04-01:
I have not done anything with the printer firmware since 3.9. It's just a matter of allowing the g-code interpreter to pass through tool values above T4, but I haven't had a reason to upgrade.
There should no be issues with any version of slicer; you just have to create a printer profile with more extruders.

Johnday29, 2022-04-01:
Awesome, build my unit as we speak!

Johnday29, 2022-04-02:
So I can't find the STL for the Idler cylinder  am I blind or is it missing?

cjbaar, 2022-04-02:
In my design, that part is the most different from the original. It's not one piece, it has to be assembled. Print one idler end, one mount, and 4x the idler-rings

cjbaar, 2024-10-18:
This project has been updated for the MMU3+Mk4. Please check the readme files for further info.

closed 2024-10-18 by cjbaar

Johnday29, 2024-10-18:
Yes! If this works I will definitely be donating to the project. Thanks.