cjbaar/prusa-mmu-12x #4: Compile MMU2S code

rmilyard, 2021-04-12:
I am trying to figure out how to compile the code.  I am not sure I understand how to do this.

Arduino: 1.8.5 (Windows 10), Board: "Original Prusa i3 MK3 Multi Material 2.0 upgrade"

Build options changed, rebuilding all
C:\Users\rmily\Desktop\MM-control-01-master\MM-control-01-master\MM-control-01\version.ino:13:10: fatal error: dirty.h: No such file or directory

 #include "dirty.h"

          ^~~~~~~~~

compilation terminated.

exit status 1
Error compiling for board Original Prusa i3 MK3 Multi Material 2.0 upgrade.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Panayiotis-git, 2021-04-12:
create a new dirty.h file (next to the main.h) with content:
#define FW_LOCAL_CHANGES 0

rmilyard, 2021-04-12:
Thanks.  I made that file and now getting this error.  I am noob when codes to compiling firmware sorry.

Arduino: 1.8.5 (Windows 10), Board: "Original Prusa i3 MK3 Multi Material 2.0 upgrade"

In file included from sketch\main.cpp:20:0:

version.h:7: error: stray '$' in program

 static const uint16_t fw_buildnr = ${GIT_PARENT_COMMITS}; //!< number of commits preceeding current HEAD

                                    ^

In file included from sketch\main.cpp:20:0:

version.h:7: error: 'GIT_PARENT_COMMITS' was not declared in this scope

 static const uint16_t fw_buildnr = ${GIT_PARENT_COMMITS}; //!< number of commits preceeding current HEAD

                                      ^~~~~~~~~~~~~~~~~~

exit status 1
stray '$' in program

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

rmilyard, 2021-04-13:
Well I have been messing with this 3 day now and just can't seem to get it compiled.

cjbaar, 2021-04-13:
So this is based on the Prusa repo, which uses Arduino to compile. It can be a lot if you are unfamiliar with Arduino and the IDE. The full build docs from https://github.com/prusa3d/MM-control-01 may help. Part of the process is you have to create your own version.h. I am using the following values. They don't mean much, but they have to be above certain versions for the printer firmware to accept communication.
```
static const uint16_t fw_version = 111; //!< example: 103 means version 1.0.3
static const uint16_t fw_buildnr = 505; //!< number of commits preceeding current HEAD

cjbaar, 2021-04-13:
My whole IDE is messed up right now because I made the mistake of updating and Arduino steps on its own libraries/avr, so I can't get 1.8.5 to work now. When I can get a working env, I can try creating a .hex file for direct upload, but right now I don't have a runtime method to alter the PFTE length.

rmilyard, 2021-04-13:
Thanks.  Right now I am getting these errors when changing per above:

Arduino: 1.8.5 (Windows 10), Board: "Original Prusa i3 MK3 Multi Material 2.0 upgrade"

Archiving built core (caching) in: C:\Users\rmily\AppData\Local\Temp\arduino_cache_507581\core\core_PrusaResearch_avr_prusa_mm_control_7808ed4679fb9bf84d9889cab7737b64.a
C:\Users\rmily\AppData\Local\Temp\ccqG87OG.ltrans0.ltrans.o: In function `tmc2130_rx':

sketch/tmc2130.c:320: undefined reference to `tmc2130_cs_low'

sketch/tmc2130.c:321: undefined reference to `spi_txrx'

sketch/tmc2130.c:322: undefined reference to `spi_txrx'

sketch/tmc2130.c:323: undefined reference to `spi_txrx'

sketch/tmc2130.c:324: undefined reference to `spi_txrx'

sketch/tmc2130.c:325: undefined reference to `spi_txrx'

sketch/tmc2130.c:326: undefined reference to `tmc2130_cs_high'

sketch/tmc2130.c:330: undefined reference to `tmc2130_cs_low'

sketch/tmc2130.c:331: undefined reference to `spi_txrx'

sketch/tmc2130.c:333: undefined reference to `spi_txrx'

sketch/tmc2130.c:334: undefined reference to `spi_txrx'

sketch/tmc2130.c:335: undefined reference to `spi_txrx'

sketch/tmc2130.c:336: undefined reference to `spi_txrx'

sketch/tmc2130.c:337: undefined reference to `tmc2130_cs_high'

C:\Users\rmily\AppData\Local\Temp\ccqG87OG.ltrans0.ltrans.o: In function `tmc2130_tx':

sketch/tmc2130.c:306: undefined reference to `tmc2130_cs_low'

sketch/tmc2130.c:307: undefined reference to `spi_txrx'

sketch/tmc2130.c:308: undefined reference to `spi_txrx'

sketch/tmc2130.c:309: undefined reference to `spi_txrx'

sketch/tmc2130.c:310: undefined reference to `spi_txrx'

sketch/tmc2130.c:311: undefined reference to `spi_txrx'

sketch/tmc2130.c:312: undefined reference to `tmc2130_cs_high'

C:\Users\rmily\AppData\Local\Temp\ccqG87OG.ltrans0.ltrans.o: In function `tmc2130_init_axis_current_normal':

sketch/tmc2130.c:218: undefined reference to `__res'

C:\Users\rmily\AppData\Local\Temp\ccqG87OG.ltrans0.ltrans.o: In function `tmc2130_init_axis_current_stealth':

sketch/tmc2130.c:205: undefined reference to `__res'

C:\Users\rmily\AppData\Local\Temp\ccqG87OG.ltrans0.ltrans.o: In function `tmc2130_init':

sketch/tmc2130.c:247: undefined reference to `selector_step_pin_init'

sketch/tmc2130.c:248: undefined reference to `pulley_step_pin_init'

sketch/tmc2130.c:249: undefined reference to `idler_step_pin_init'

sketch/tmc2130.c:250: undefined reference to `selector_step_pin_reset'

sketch/tmc2130.c:251: undefined reference to `pulley_step_pin_reset'

sketch/tmc2130.c:252: undefined reference to `idler_step_pin_reset'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Original Prusa i3 MK3 Multi Material 2.0 upgrade.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

cjbaar, 2021-04-13:
Yeah I am getting that, too. These files come straight from the original, and I haven't touched them since last successful compile. I think it's because, like I said, I updated the IDE, and now the AVR is showing as v7.3.0, and probably has a different core now. Not sure how to revert the back-end without wipe and re-install.

cjbaar, 2021-04-13:
Yeah, this is really frustrating. I can get it to compile on an older laptop with an untouched install of 1.8.5, which is using avr-gcc v5.4.0. Unfortunately, IMO, the Ardiuno IDE does not deal with upgrades and libraries/sketches that compiled fine under certain avr versions. I still have a copy of 1.6.11 because I have a sketch that breaks serial communication with any other version.

cjbaar, 2021-04-13:
I add compiled .hex files to this repo. However, I *have not tested them*. Use at your own risk.

rmilyard, 2021-04-13:
Ok.  I have all part pretty much printed.  Still trying to figure out how all goes together but hope to test soon!

rmilyard, 2021-04-14:
Do you have anymore build info?  I am trying to figure out how mounted idler in the upper part.

cjbaar, 2021-04-14:
Currently I don't; actually wasn't expecting this to catch on. :) It mostly assumes you previously built the Prusa model, and most of the design is similar. For the idler, you assemble the 14 pieces on a 5mm shaft. Repeat the 1/2/3 pattern (# printed on bottom), and you can optionally embed m3 screws to keep it tight as you go (every 3 rings). The oval piece holds the bearing, and mounts to the top with two m3 screws.

rmilyard, 2021-04-15:
I have few questions maybe if don’t mind can email me so can get this built. I am willing to help any way I can.   My email is rmilyard@hotmail.com

cjbaar, 2021-04-15:
Sorry @rmilyard, looks like hotmail is blocking IP ranges from my mail host. Do you have another address? Otherwise, ask here and I will do what I can to help.

rmilyard, 2021-04-15:
rmilyard@gmail.com and rmilyard@cox.net

cjbaar, 2021-04-15:
@rmilyard I added a few more pics under [https://github.com/cjbaar/prusa-mmu-12x/blob/master/TIPS.md](https://github.com/cjbaar/prusa-mmu-12x/blob/master/TIPS.md) with some additional assembly tips. Hope this helps.

rmilyard, 2021-04-15:
Ah. So like this?

[cid:08121F0B-0515-488F-BCD7-74189DE612AB]

[cid:82D47372-7244-432D-B3FE-30042E4BFD3B]

cjbaar, 2021-04-16:
Sorry I don't see any images with that post; just some sort of id ref.

cjbaar, 2021-04-19:
@rmilyard I did upload some compiled .hex files, but based on my testing, PrusaSlic3r will not uploaded it properly. It is not recognizing it as being for the MMU, and so is trying to use the avr settings for the einsy instead. I am not sure what PrusaSlic3r uses to determine this.

cjbaar, 2021-04-19:
@rmilyard OK; it helps to read the docs. :) I had to manually add a comment to the beginning of the hex files to be recognized as being for the MMU. I did some brief testing, and was able to upload the included .hex files to the MMU using PrusaSlic3r.

flinke-flasche, 2021-04-23:
That sounds good. Did you print a test if the MMU works well with your firmware?

Whats about the firmware for the MK3+? did it work as well?

zonkie, 2021-06-11:
I flashed your Files yesterday and assembled everything, but as soon as the printer wants to start Printing, it Resets itself. After the reset, the printer does the bed leveling procedure, and the MMU cries that it has a homing error, because the Filament ist still loaded... Did i mess up the wiring there, or is this some kind of Safety feature in the Printers firmware?

zonkie, 2021-06-11:
I think, this ist some security mechanism of the printer. It checks, is the file was sliced for the Pruša and if not, it resets.

Also the idler positions itself at the wrong direction, for me. But i have to wait for two u-bearings anyway, gonna disassemble it again then ;(

cjbaar, 2021-06-13:
Don't think I have run into the reset issue before. Are you using something other than Prusa Slic3r for slicing? Not that the printer should care.
Can you describe more what you mean by "the idler positions itself at the wrong direction?" As I mentioned in the notes, I tried using the motor feedback for homing, but it doesn't always work well. If it is not resetting all the way back, you could try manually moving it to the end before turning on.

zonkie, 2021-06-13:
Yes i used m existing prusa profiles on prusaslicer (only added some extruders :) ). I also created a custom profile with the settings from the original one.
I read in the Prusa Forums that this might be a mechanism that happens if the code was sliced for a different printer...

For the idler positioning... well, it seems, that when starting the print, it first selects the correct feed line, but before loading, it turns the idler from e.G. Position 3 to 2, but the output selector stays at 3. I can see the dot for number 2 at the top slit, so it seems not to happen accidental but is precisely positioned. I'm still waiting for two more u-bearings, so i have to take it apart again anyway. I will see of i can adjust something then.

Also ill try to flash all firmwares again, just to rule out that i have flashed maybe the original firmware or so...

zonkie, 2021-06-14:
So here is another case in the prusa forum, he is also having the issue with the wrong slot loaded in the MMU. So it seems to be an issue with the original mmu and mk2s firmwares too. I'll try to adjust the filament probes as soon as i have time for it, as this might be the issue in some cases there.

https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-assembly-and-first-prints-troubleshooting/i3-mk3s-mmu2s-rebooting-at-print-start/#post-173558

BTW: Is your Firmware compiled for mk3 or mk3s? As i'm using the MK3s That could be be an issue too. In that case i might have to try and port your changes to the latest MK3s firmware :D :S

flinke-flasche, 2021-06-14:
I use the MMU12x with my MK3s without issues. 

What do you mean with wrong load into wrong slot?
The selector stands on filament 5 but the idler stands at filament 6 and is moving it?

zonkie, 2021-06-14:
Yes that's what i meant. 

I calibrated the F.I.N.D.A probe now, and i could start a print (in Single mode for now). In the end it seems like it was an error 15 - "Error is sitting in front of the computer"... *woops, i'm Sorry* ;)

I think this "jumping to a different feed-line" was just some Result of the printer panicking in the Reset process...

flinke-flasche, 2021-06-14:
@zonkie 
i had the same issue that the selector stands on filament 5 and the idler stands at filament 6 too.
When the selector moved to filament 1 and the idler stands at filament 2 i've pressed the button again. The idler moved to the first position. Since that the MMU12x works perfectly.

zonkie, 2021-06-14:
So my first print would have been mostly successful, if i didn't skew the x gantry by hand whilst turning tje motors manually :S 

But the issue with the wrong selected filament feed lines persists. In my test print, Stine of the squares have the same color, but should be different... Green and pink are double

Can't upload images, so here is a link to my Twitter post :D
https://twitter.com/WorsMitVitamine/status/1404523338203009030?s=19

cjbaar, 2021-06-15:
I guess I'm not entirely clear on the selector issue from the descriptions above. Are you saying the selector isn't moving on a filament change request? With the FINDA so far back, it shouldn't be getting jammed. Is the selector grinding when moving?

zonkie, 2021-06-16:
Hi, sorry i'm answering so late. 

Well, the issue is, that the MMU starts to load the correct filament, but before pushing it in, it moves both the idler as well as the selector one Position in one direction or the other.

I couldn't observe it often enough, to find a pattern there but in the Prusa forums link i posted earlier, this issue seems to have happened with the original Firmware, too...

cjbaar, 2021-06-18:
OK, I see. So, the MMU is fairly "dumb" as compared to the printer. The MMU doesn't do anything unless the printer sends a command, so if the selector/idler are moving, it is because the printer asked for another filament. Based on what you've said, I think it is most likely the runout feature, which tells the MMU to pick the next filament if the printer thinks it has run out. This would be triggered if the FINDA on the MMU selector, so if it is happening when the filament is still loaded, then the FINDA probably needs to be adjusted down a bit. I might also check to make sure the filament isn't being ground out too much somewhere in the load process, and therefore has a gap that accidentally triggers the FINDA to think it is empty.

I believe you can also just disable the runout feature somewhere in the printer menus if you don't need that. I think it was more designed for if you use the MMU as a constant feeder instead of a color-changer.

zonkie, 2021-06-18:
Yes, since i adjusted the finda probe, i don't have these issues anymore. i think, the shaking from the load process lead to misfires in the sensor :) 

Right now, i'm trying to implement the 12x changes in the 3.10.0 firmware, but i can't compile it yet. I hope, i'll find some spare time in the next days.

cjbaar, 2021-06-18:
Cool; let me know how that goes. I haven't gotten around to looking at 3.10 yet. If you don't use the menus to "Load Filament X" or need to auto-deplete to work correctly, the only required change to get above 5 is in Marlin.cpp, iirc

Sdee619, 2023-08-19:
Hello I put together your mmu12 design but I am having trouble understanding how to generate a .hex file for the mmu2
I have a genuine mmu2s I converted into a mm12 with all the parts put together. I cant make the .hex file in arduino I followed all the steps but Im new to coding and arduino I have little experience. I opened mm- control-1 in arduino added the version.h file and dirty.h with the information required. Under config mm-options I added prusa-mmu2s-display since im just using the mini display. when I try to verify I get this error code and dont know I am doing wrong

C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.cpp: In function 'void display_test()':
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.cpp:62:37: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
     display_message(MSG_INITIALIZING);
                                     ^
In file included from C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.cpp:1:0:
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.h:59:13: note:   initializing argument 1 of 'void display_message(char*)'
 extern void display_message(char *msg);
             ^~~~~~~~~~~~~~~
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.cpp: In function 'void display_command(char, uint8_t, boolean)':
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.cpp:117:62: error: default argument given for parameter 3 of 'void display_command(char, uint8_t, boolean)' [-fpermissive]
   void display_command(char c, uint8_t v, boolean force=false) {
                                                              ^
In file included from C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.cpp:1:0:
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.h:55:13: note: previous specification in 'void display_command(char, uint8_t, boolean)' here
 extern void display_command(char c, uint8_t v, boolean force=false);
             ^~~~~~~~~~~~~~~
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp: In function 'void motion_set_idler_selector(uint8_t, uint8_t)':
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp:77:34: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
     display_message(MSG_SELECTING);
                                  ^
In file included from C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp:14:0:
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.h:59:13: note:   initializing argument 1 of 'void display_message(char*)'
 extern void display_message(char *msg);
             ^~~~~~~~~~~~~~~
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp:96:29: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
     display_message(MSG_IDLE);
                             ^
In file included from C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp:14:0:
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.h:59:13: note:   initializing argument 1 of 'void display_message(char*)'
 extern void display_message(char *msg);
             ^~~~~~~~~~~~~~~
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp: In function 'void unload_to_finda()':
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp:134:34: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
     display_message(MSG_UNLOADING);
                                  ^
In file included from C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp:14:0:
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.h:59:13: note:   initializing argument 1 of 'void display_message(char*)'
 extern void display_message(char *msg);
             ^~~~~~~~~~~~~~~
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp: In function 'void motion_feed_to_bondtech()':
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp:166:32: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
     display_message(MSG_LOADING);
                                ^
In file included from C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp:14:0:
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.h:59:13: note:   initializing argument 1 of 'void display_message(char*)'
 extern void display_message(char *msg);
             ^~~~~~~~~~~~~~~
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\motion.cpp:180:40: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
           display_error(MSG_LOADING, tr);
[... cut ...]
                                 ^
In file included from C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\mmctl.cpp:18:0:
C:\Users\Streaming PC\Documents\MM-control-01-master (3)\MM-control-01-master\MM-control-01\display.h:59:13: note:   initializing argument 1 of 'void display_message(char*)'
 extern void display_message(char *msg);
             ^~~~~~~~~~~~~~~

exit status 1

Compilation error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

cjbaar, 2023-08-20:
@Sdee619 Yeah, I get the same when compiling under Arduino IDE v2. It looks like something has changed in the strict checking of types; maybe why Prusa has unfortunately moved away from the IDE for the newer firmware. The last time I compiled all of this was under v1.8.5, I think. I would try downloading the older version of the IDE and see if it works for you.

Sdee619, 2023-08-20:
I now tried 1.85 and I am receiving this error code
Using library Wire at version 1.0 in folder: C:\Users\Streaming PC\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.5\libraries\Wire 
Using library SSD1306Ascii at version 1.3.5 in folder: C:\Users\Streaming PC\Documents\Arduino\libraries\SSD1306Ascii 
Using library SPI at version 1.0 in folder: C:\Users\Streaming PC\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.5\libraries\SPI 
exit status 1
Error compiling for board Original Prusa i3 MK3 Multi Material 2.0 upgrade.

Sorry I am new to this I'm just wondering If you were to make a .hex file why can't we all use the same one? is there a way just to upload the .hex file here for me test?

Sdee619, 2023-08-20:
I will try Version 2 again Hiro on twitter made your mmu12 and said Also, when compiling with Arduino IDE ver.2, there was an error about the code notation. After correcting them one by one, I was able to compile. So I will try this now

cjbaar, 2023-08-21:
That output from 1.85 doesn't show any actual errors, so it's hard to say what the issue is. There are pre-compiled hex files in the firmware folder of the repo, but I can't guarantee it has the right PTFE length for your setup. My firmware does not have the calibration menu for length.

Sdee619, 2023-08-21:
Ok I will try it still I ended up getting this error code if I will try and seek help and see why this error code happens I saw someone had the exact error in previous messages 

C:\Users\STREAM~1\AppData\Local\Temp\ccjNxywj.ltrans0.ltrans.o: In function `tmc2130_rx':
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:320: undefined reference to `tmc2130_cs_low'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:321: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:322: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:323: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:324: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:325: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:326: undefined reference to `tmc2130_cs_high'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:330: undefined reference to `tmc2130_cs_low'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:331: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:333: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:334: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:335: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:336: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:337: undefined reference to `tmc2130_cs_high'
C:\Users\STREAM~1\AppData\Local\Temp\ccjNxywj.ltrans0.ltrans.o: In function `tmc2130_tx':
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:306: undefined reference to `tmc2130_cs_low'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:307: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:308: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:309: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:310: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:311: undefined reference to `spi_txrx'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:312: undefined reference to `tmc2130_cs_high'
C:\Users\STREAM~1\AppData\Local\Temp\ccjNxywj.ltrans0.ltrans.o: In function `tmc2130_init_axis_current_normal':
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:218: undefined reference to `__res'
C:\Users\STREAM~1\AppData\Local\Temp\ccjNxywj.ltrans0.ltrans.o: In function `tmc2130_init_axis_current_stealth':
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:205: undefined reference to `__res'
C:\Users\STREAM~1\AppData\Local\Temp\ccjNxywj.ltrans0.ltrans.o: In function `tmc2130_init':
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:247: undefined reference to `selector_step_pin_init'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:248: undefined reference to `pulley_step_pin_init'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:249: undefined reference to `idler_step_pin_init'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:250: undefined reference to `selector_step_pin_reset'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:251: undefined reference to `pulley_step_pin_reset'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/tmc2130.c:252: undefined reference to `idler_step_pin_reset'
C:\Users\STREAM~1\AppData\Local\Temp\ccjNxywj.ltrans0.ltrans.o: In function `enhanced_interactive_menu()':
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/mmctl.cpp:462: undefined reference to `display_menu_options(char const*, char const*, char const*)'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/mmctl.cpp:545: undefined reference to `display_menu_options(char const*, char const*, char const*)'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/mmctl.cpp:499: undefined reference to `display_menu_options(char const*, char const*, char const*)'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/mmctl.cpp:524: undefined reference to `display_menu_options(char const*, char const*, char const*)'
C:\Users\Streaming PC\Documents\MM-control-01-master\MM-control-01-master\MM-control-01/mmctl.cpp:564: undefined reference to `display_menu_options(char const*, char const*, char const*)'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

cjbaar, 2024-10-18:
Closing this, as the current project has migrated to MMU3+Mk4, which does not use the old compiler process.

closed 2024-10-18 by cjbaar