cjbaar/prusa-mmu-12x #7: Powered on.

rmilyard, 2021-05-06:
So I plugged it all in as would MMU2S.  I flashed both firmware.  I am not seeing it.  I see light on MMU2 board but MK3S not seeing MMU2 connnected.

rmilyard, 2021-05-11:
Nope never got working

cjbaar, 2021-05-13:
Sorry I don't have access to the boards right now. I tried the firmware I uploaded, and have the same issue. The MMU boots, but printer does not see it. Not sure what needs to be changed, but for the moment I am afraid that you will need to be able to compile this locally on Arduino 1.8.5 to upload.

rmilyard, 2021-05-13:
That sucks!  I have it completely built but like said isn't seen my printer.

flinke-flasche, 2021-05-14:
@rmilyard 
Have you tried to print something with the new firmware without the MMU2? Does it work?

@cjbaar 
Maybe something is missing in one of the firmwares? 
The MK3s have to know that a MMU2 is connected and the MMU2 have to say “hello MK3s, here i am. Please unlock the menu.“.

At the beginning of printing with the MMU2, i had the same issue. The MK3s didn't saw my MMU2 several times.
This was before the 1.6 firmware was launched.
I had to resart the MK3s several times till it recognize the MMU2.

rmilyard, 2021-05-15:
Got it all loaded.  I will play with it more tomorrow.  How tight use have idler screws?

flinke-flasche, 2021-05-18:
@rmilyard 
any news yet?

My last parts arrived today. now i'm ready the transform my MMU2 in a MMU12x :-)

rmilyard, 2021-05-18:
@flinke-flasche  I was able to get it to boot up.  I haven't had time to mess with it.  First try I couldn't get filament to load yet.

flinke-flasche, 2021-05-22:
I set up my MMU12x. Everything works perfectly. I love it and it looks much better as the MMU2 :-)

@rmilyard do you think you can compile a multi language Firmware for the MK3s? i'm from Germany :-)

I use Octoprint for my Printers. Octoprint can't use only 10 Extruders. i do a feature request to Octoprint.

cjbaar, 2021-05-26:
The MK3 firmware includes files for multi-language compilation, though I have not tried using them before. Unfortunately, at the moment, I have not put similar adjustments in the MMU display code.

As far as octoprint limiting filaments, I had similar issue with Prusa, which is why I had to modify the MK3 firmware as well. Not sure why these systems have so much hard-coding; it should literally just be forwarding on whatever you find in the g-code.

cjbaar, 2021-05-26:
@rmilyard I only tighten my idler screws a couple of turns. I don't think it needs as much pressure as the original, but your results may vary.

flinke-flasche, 2021-05-30:
@cjbaar 
do you think you can compile the Firmware with multilanguage?

Maybe you can add a post at my Octoprint request.
https://github.com/OctoPrint/OctoPrint/issues/4149

Can you tell me what the tiny numbers under the selected filament at the display is showing?

cjbaar, 2021-06-03:
1) I don't have a working Arduino IDE 1.8.5 right now. I need to get something working again to compile these.
2) What is the issue with OctoPrint? I haven't used it much. Is it only sending G0-G9, so the last two filaments end up repeating filament color #10?
3) The numbers show total load/unload attempts and failures. Describe on the FW page: https://github.com/cjbaar/MM-control-01

zonkie, 2021-06-04:
I think it's just the Max-Value in the HTML5 Field in the template. I could just put in 12 and send it, and it saved :)

I haven't got my MMU Working as i'm waiting for Parts, so i couldn't try it out, but otherwise i'd open a pull request for that change in OctoPrint and also in the MMU2S Filament Selector Plugin.

flinke-flasche, 2021-06-06:
@zonkie 
You’re right. I can type it manually in the field and then it works perfectly.
I’ve tested it today.

bist du auch aus Deutschland?

@cjbaar 
No more octoprint problems. 
thank you for the display description.

i love your MMU12x. 
thank you for sharing.
The firmware works perfectly for me.

zonkie, 2021-06-06:
I'm still waiting for parts... An m3 screw set and 10 of the 12 ordered u-shaped-bearings have arrived yet (thanks, ebay-seller, the bag is even labeled with "12pcs"...) Also some of the AliExpress things seem to have arrived in Germany, waiting for customs :) also i'm printing lots of spool-holders...

@flinke-flasche yes, i live near Hamburg :)

closed 2024-10-18 by cjbaar