cjbaar/prusa-mmu-12x #8: Stuck in tool change 1

Imaginex44, 2021-07-05:
Installed and updated both firmwares. The print is stuck on tool change 1 and just freezes over the purge block. Running MMU board firmware without screen.

zonkie, 2021-07-06:
Try helping it with the filament change, for me it was usually that it didn't recognize the filamemt was ejected, when this error occured.

Obel1sk, 2021-07-06:
I use a lot of PETG and can get a lot of stringing until I get the temps dialed in for each roll. When the display reports an UnLoad failure, I unscrew the PTFE connector and push the filament out the front of the selector*, clip the string and manually position the filament back to the load position**, press "OK" on the display interface and the MMU will re-cycle and re-detect the filament, clearing the error and completing the unload.

* I use the "original m5 heat-set remix" so as not to wear out the plastic threads.
** replace the PTFE connector

cjbaar, 2021-07-06:
If you compiled without the screen, then the MMU LEDs should serve as indicators just as they did before, just any filament over #5 will show as LED #5, I believe. Without more info, I am guessing one of the filament sensors needs alignment. What does the printer show? Is the extruder motor still moving? Did the filament unload? Not sure what "stuck" means.

Prusaprinter44, 2021-07-11:
The filaments all load to nozzle and it appears the issue is happening at
the bondtech area in the extruder, it retracts to just above the gear and
the mmu itself is not retracting the filament beyond that point.  All of
the sensors are working, I am thinking it may be in the U channel bearing
as i have noticed there are 2 versions of bearings, 1 with a deeper groove
and 1 with a less pronounced groove in the bearing, i have tried both
versions with the same result, can anyone send some screen shots of the
tension springs? It seems much different then the regular mmu, we currently
run 12 five color mmus without any issues so tuning this shouldn't be a
problem.

flinke-flasche, 2021-07-12:
How long is your PTFE tube?
Did you calibrate the PTFE length?

I had problems after I calibrated the PTFE length. 
After resetting the MMU and flashing the firmware everything works fine.

Prusaprinter44, 2021-07-13:
I have gone back and forth between the stock length and a few mms longer, This may sound silly but does this setup require the Bowden tube calibration like it did in the past? My current length on the bowden tube is 14.17 " or 359 mm.  Using a m6 fitting on the selector and m10 on the IR sensor.

I think I have narrowed it down to spring tension on top or a bowden length issue. All of the commands work fine manually, the problem occurs when it wants to unload.

cjbaar, 2021-07-14:
So in the interest of space, and my lack of understanding of Prusa's use of eeprom storage, my code disabled all of the length calibration. You just need to enter the length (in mm) in your config.h when compiling if it's not the same as the stock Prusa PTFE. I doubt any of that would affect MMU functions though. On load, the MMU will stop as soon as the filament hits the extruder, you just might get a bit of grinding if it is set too short. If it is set too long, the printer will ask for "C0" (continue) so the filament would eventually get there anyway. During unload, the length only comes into play during the last part (as it approaches the FINDA). If the MMU isn't retracting, it's because the printer is not telling it to. The MMU is pretty dumb in comparison to the printer firmware, and generally only does what it's told. The screen would help debug (as it outputs the last command received), but if it not making any attempt to retract, then I would really lean toward the filament sensor on the extruder being misaligned.

closed 2024-10-18 by cjbaar