cjbaar/prusa-mmu-12x #10: Firmware compiling

flinke-flasche, 2021-08-21:
Does anyone have compile the actual firmware for using the Super PINDA?

cjbaar, 2021-08-21:
I wasn't aware they changed the sensor. It looks like it says it requires 3.9.2 or higher, and my mmux12 was forked at 3.9.0. Since they don't seem to have any interest in merging my changes, the mods will have to be re-done. I might try to work on a patch for future version, but am not able to do it for a while.

flinke-flasche, 2021-08-25:
The MK3S+ got this new sensor.
It's not really necessary right now. I can install it on my second prusa. 
But if the sensor is really better than the old one, i want to install it on the prusa with the mmu12x.

flinke-flasche, 2021-10-30:
Thank you.
Works fine.

VSCustoms, 2021-11-03:
Hi corn62. 

 I am still fairly new to VSCode so I have not completely figured out where to make all the changes. Would you be able to compile it for a MK2.5s with RAMBo13a board?  

 I cant seem to find where I make the changes in the Marlin_main.cpp area.

corn62, 2021-11-03:
To be fair I am far from being an expert myself.
I just went down the changes initially made by the author (https://github.com/prusa3d/Prusa-Firmware/compare/MK3...cjbaar:MK3) and apply them all to the latest Prusa Firmware available on Github.
Then compiled everything following Prusa's guidelines.

Both files (3.9.0 and 3.10.0) were almost the same, I have no idea what it would look like for the MK2.5s.

VSCustoms, 2022-12-21:
> 

I finally got a MK3S+ to try your firmwear out on. When loaded does it say MMU NA? did you happen to update yours to the new firmwear? just trying to get it all set up before I test print.

cjbaar, 2022-12-22:
@VSCustoms No, I haven't touched the printer firmware since 3.9.something. I think you have to set some version vars in one of the config files of the MMU code. It should not say "N/A" it should show the version connected.

cjbaar, 2022-12-31:
I have updated release 3.11.0 with the necessary changes for tools above T4. It is on my fork of PrusaFirmware, and there is a compiled MK3S version in the firmware folder of this repo.

closed 2022-12-31 by cjbaar