cjbaar/prusa-mmu-12x #11: Custom Firmware for MK2.5s?

VSCustoms, 2021-08-24:
I’m just wondering if anyone has done custom firmware for the MK2.5s? I ordered everything to do this and then realized the firmware was just MK3s.

VSCustoms, 2021-11-01:
Ok I seem to be a little lost as I am unsure what I need to change now. I`m running 3.10.0 but I need your mods compiled for a MK2.5S vs a MK3S. Just not exactly sure what I`m changing. I`m still learning VScode.

VSCustoms, 2021-11-01:
Also I’m not worried about single mode settings as I never slice using that. I just pick my colour and if it’s only one I do that and print. I just need to be able to run 12 colours VS 5 if at all possible.

cjbaar, 2021-11-01:
Here are the changes I made in 3.9. You don't really need the AutoDeplete stuff if you don't use it, but you need the change in Marlin.cpp that recognizes codes above T4, and some of the stuff in mmu.h/mmu.cpp that relays those codes. The lcd changes make things print prettier, but may not be necessary. It's mostly about getting around the fact that they hard-coded only T0-T4 as acceptable values.

VSCustoms, 2021-11-23:
Ok I have finally figured out how to make it for a MK2.5s but I'm getting an error in the ultralcd.cpp file in arduino saying

 invalid conversion from 'const char*' to 'char' -fpermissive

closed 2021-12-15 by VSCustoms