cjbaar/prusa-mmu-12x #12: Compiling issue

VSCustoms, 2021-12-14:
Ok I got firmware 3.9.0 to compile and made a .hex file for my MK2.5s

My issue is when I redid everything to try and get 3.10.0 to work I get this .

C:\Users\Tim\AppData\Local\Temp\ccy70zps.ltrans0.ltrans.o: In function `process_commands()':

ccy70zps.ltrans0.o:(.text+0x5040): undefined reference to `mmu_filament_count'

C:\Users\Tim\AppData\Local\Temp\ccy70zps.ltrans25.ltrans.o: In function `Config_StoreSettings()':

ccy70zps.ltrans25.o:(.text+0x588): undefined reference to `mmu_filament_count'

C:\Users\Tim\AppData\Local\Temp\ccy70zps.ltrans25.ltrans.o: In function `ad_getAlternative(unsigned char)':

ccy70zps.ltrans25.o:(.text.unlikely+0x23e): undefined reference to `mmu_filament_count'

C:\Users\Tim\AppData\Local\Temp\ccy70zps.ltrans5.ltrans.o: In function `mmu_loop()':

ccy70zps.ltrans5.o:(.text+0x10ae): undefined reference to `mmu_filament_count'

ccy70zps.ltrans5.o:(.text+0x12e6): undefined reference to `mmu_filament_count'

C:\Users\Tim\AppData\Local\Temp\ccy70zps.ltrans19.ltrans.o:C:\Users\Tim\AppData\Local\Temp\arduino_build_755722\sketch/ultralcd.cpp:6240: more undefined references to `mmu_filament_count' follow

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board PrusaResearch EINSY RAMBo.

Prusaprinter44, 2022-03-23:
Did you ever get this up and running on the 2.5s. I'm giving this a go again on a few of my machines.

VSCustoms, 2022-12-08:
> 

Not sure how I missed this message. Unfortunately I did not get it working. It fought me the whole way and I needed my MMU up and running. I just recently got a MK3S+ I will be tackling it again on as I really would love to have the 12 colour option and I have been playing with VS code more maybe the 2.5S board did not like the custom farmwear.

cjbaar, 2024-10-18:
Closing this, as this project has moved on to MMU3+Mk4, which does not use the old compiler process.

closed 2024-10-18 by cjbaar