cjbaar/prusa-mmu-12x #16: MMU FW UPDATE NEEDED

grzlus, 2023-01-14:
I built and flashed both MMU and printer. For printer I used my own compiled software, as I have installed Mosquito Magnum with SliceEngeenering thermistor. Now I'm getting error `MMU FW UPDATE NEEDED` and in serial I communication I see:

``` bash
Recv: echo:MMU2:>X0*4c.
Recv: echo:MMU2:>S0*c6.
Recv: T:19.8 /0.0 B:23.0 /0.0 T0:19.8 /0.0 @:0 B@:0 P:24.8 A:28.7
Recv: echo:MMU2:Version mismatch, last bytes: 6b 6f 6b 6f 6b 6f 6b 6f 6b 6f 74 73 00 00 00 00
Recv: echo:MMU2:MMU FW UPDATE NEEDED
Recv: retryAttempts=3
Recv: echo:MMU2:<stokokokokok.
```

Do you know how I can check what is sent via MMU to my MK3S to debug it? I tried adding some code to debug, but I did not see anything in serial communication:

``` diff
--- a/Firmware/mmu2_protocol_logic.cpp
+++ b/Firmware/mmu2_protocol_logic.cpp
@@ -213,6 +213,11 @@ StepStatus ProtocolLogic::ProcessVersionResponse(uint8_t stage) {
         SendVersion(stage);
     } else {
         mmuFwVersion[stage] = rsp.paramValue;
+        MMU2_ERROR_MSGRPGM(PSTR("Version test"));
+        SERIAL_ECHOPGM("Stage: ");
+        SERIAL_ECHOLN(stage);
+        SERIAL_ECHOPGM("Number: ");
+        SERIAL_ECHOLN(mmuFwVersion[stage]);
```

I'll try to find some old thermistor and connect it to test your precompiled firmware to rule out fired chip for communication.

cjbaar, 2023-01-14:
I have not tested this with non-prusa firmware. I am guessing it want something specific from the MMU, so you would need to disable that in your printer firmware, or compile a version of the MMU firmware that gives it the version it wants, but I can't guarantee that would work.

grzlus, 2023-01-14:
But I forked your repository and only changed some values to work with thermistor.

And my changes should not affect MMU communication.

cjbaar, 2023-01-15:
What do you have in `version.h` for the MMU? I use the following:
```
static const uint16_t fw_version = 111; //!< example: 103 means version 1.0.3
static const uint16_t fw_buildnr = 505; //!< number of commits preceeding current HEAD
```
My guess is the printer wants at least 105 for `fw_version`, possibly 106.

grzlus, 2023-01-16:
I did not built MMU firmware, but I'll do that. It's a good starting point and I'll try to find the cause there.

grzlus, 2023-04-01:
So... I was using wrong branch on the other repo. Now It's fine. I had some problems with different printed parts, but overall it's a great project. I'd like to check how MMU3 looks like and adapt it to that, at least the 12 colors part :)

When I'll have time, I'll update my MK3 to MK3.5 and I'll begin testing.

closed 2023-04-01 by grzlus