cjbaar/prusa-mmu-12x #17: mmu12x-lower-mod.stl is damaged

renoxray, 2023-03-31:
Hi @cjbaar,

I finally trying to build one mu 2x based on my exist mmu2s and I found the mmu12x-lower-mod.stl come with only 78.6k size.

Could you help to update it :)

[image]

grzlus, 2023-04-01:
This looks like a modifier to the file `mmu12x-lower.stl`. You can use it to disable or enforce supports

renoxray, 2023-04-02:
@grzlus 

Thanks for the help. I've just realized the mod part is for this purpose.

[image]

closed 2023-04-02 by renoxray