cjbaar/prusa-mmu-12x #18: MMU-12x stuck in "Initializing"

Agent-Cain, 2023-07-14:
I've build the MMU-12x and flashed the firmware with display on it. 
Display turned on instantly.
Also flashed the 3.11. FW on the printer, no problem here. 
Printer shows under "support" MMU2 N/A, not sure if this is right. 

Now if i start up the printer (with the mmu) the LEDs flash from right to left, none stays on. 
The display shows Initializing and none of the buttons does anything. 
When I unplug the connector from the mmu to the printer nothing turns on, so I'm relatively sure that the wiring is correct. 

What am I missing 🤔

Hope someone is still active here, thanks in advance !

Agent-Cain, 2023-07-15:
After some fiddeling around I found some things out: 
Normal FW on Printer and MMU2 will work together, so connection is ok. 
Custom cjbaar FW on Printer with stock FW on MMU establishes also a connection and works so far. 
Custom cjbaar FW on printer and cjbaar FW with display on MMU doesn't work. Seems like Printer and MMU do not establish a connection. 

But the cjbaar FW on Printer and the cjbaar FW without display on the MMU are working together, and the first print with it is runnung right now. 

So there seems to be a problem with the display version of the FW, in my case it doesn't establish a connection and is stuck in Initializing and won't respond to any button presses. Someone else out there had the same problem and found a work around ?! 
Would appreciate it if you would let me know, thanks :)

grzlus, 2023-07-31:
Yeah, I had some problems, but it was only with custom built, as I needed things for Mosquito Magnum, firmware by myself. I fixed it by using correct branch.

cjbaar, 2023-08-20:
Sorry, I don't have a good answer for you. I'm just looking through the routines it runs through in startup. After initializing, it should try to home I think, but I am not sure I see where it would get hung up there.

Agent-Cain, 2023-10-27:
Oh dang, sorry for the late response. 
It doesn't even home 😅

grzlus, 2023-10-27:
Start with the [fix](https://www.printables.com/model/620818-mmu2-5-volt-power-fix?fbclid=IwAR3sfG-mzWHs7wWj6MpnYlzBHReKb1TUMogTrkOFsHoj7DBSxSszwmZHasc) as it might be the case of your problems with homing. On x12 hardware I had a problem with frequent restarts. More frequent than with original one. At least this will eliminate most of your problems.

I installed some DC-DC StepDown to 5V and I'm now having stable 5.2V on the line and no restarts at all.

Agent-Cain, 2023-12-29:
> Start with the [fix](https://www.printables.com/model/620818-mmu2-5-volt-power-fix?fbclid=IwAR3sfG-mzWHs7wWj6MpnYlzBHReKb1TUMogTrkOFsHoj7DBSxSszwmZHasc) as it might be the case of your problems with homing. On x12 hardware I had a problem with frequent restarts. More frequent than with original one. At least this will eliminate most of your problems.
> 
> I installed some DC-DC StepDown to 5V and I'm now having stable 5.2V on the line and no restarts at all.
> 
> ![IMG_0179](https://private-user-images.githubusercontent.com/1580215/278602301-7ac19528-af3f-45f0-b7aa-ebafc4195e70.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM4NTI0NDUsIm5iZiI6MTcwMzg1MjE0NSwicGF0aCI6Ii8xNTgwMjE1LzI3ODYwMjMwMS03YWMxOTUyOC1hZjNmLTQ1ZjAtYjdhYS1lYmFmYzQxOTVlNzAuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyMzEyMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI5VDEyMTU0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgxNzVjNmZjYzRhNWIzN2JkZjA2NDUzZGJlMDNiNDBmNzBhZWI3N2EwN2MzYjc0ZWRkMzcxZDhmYmJhM2FlNGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pIL8M6c972s41M-5dXDUo0JbG1dhsG7jm4yoX7mWYQU)

That was way over the top I wanted to do with my board :D 
I got some prints out of the 12 color unit without any trouble (running the non display verison). 
Today I received my PD Board from Aliexpress, which seems to be a lower hanging fruit. 
After I've installed that I'll give the display version another try!

Agent-Cain, 2023-12-29:
So, even the PD-Board didn't change a thing of my problem. 
Still no movement on a button press or anything., still flashing lights from right to the left but none staying on. 
[image]
Just stuck in this stage, no operation possible.
Now that I familiarized with the MMU and their operation I noticed something more: 
With the display version of the software the menu on the Prusa doesn't show certain options like "Load Filament" or "Load Filament to the Nozzle". 
As I have printed already a few things on that MMU I'm sure everything is working and correctly wired.

So to me it seems like the MMU gets stuck somewhere, is there a possibility to debug it somehow ?
I really would love to get this version with the display up, every kind of help is welcome :) 

Thanks in advance

flinke-flasche, 2023-12-29:
Does your MMU2 work with the firmware without display?

flinke-flasche, 2023-12-29:
I try a fix for the 5V problem as well, because my selector stucks.
I follow the guide in Chris' YouTube video. 

I had an issue with the cable I soldered on my first try.
At this time my mmu2 didn't work. When i unpluged my cable and tried it again it worked again. 

Maybe its a problem with the cable to your display?

Agent-Cain, 2023-12-29:
> I try a fix for the 5V problem as well, because my selector stucks. I follow the guide in Chris' YouTube video.
>
> I had an issue with the cable I soldered on my first try. At this time my mmu2 didn't [...]

As the display works (kind of with the display fw) but the mmu not I would guess its not a cable related problem.