cjbaar/prusa-mmu-12x #19: mmu3

Dutchy1969, 2023-07-30:
any chance we get a fork of the mmu3 firmware and i think 3.13 for the mk3s+ and mk4 support?

i was hoping to get rid of my mk3s+ and move it over to the mk4 i started looking at the firmware and stuff but im not smart enough for that.

grzlus, 2023-07-31:
I was thinking on porting changes here to TZB firmware, but porting it MMU3 might be better idea.

I want to buy MK4 upgrade too (MK3.5 probably), but I need to find more free time for that 😀

cjbaar, 2023-07-31:
I have not yet dug into the code, but the MMU3 is a complete re-write, with a total overhaul of the communication protocol. I am guessing it won't make much sense to move my changes into it, because I did a lot of re-writing of the MMU2 code. Also, they have not yet announced what the "Mk4" version of the MMU3 will be. It is my understanding that the printer itself is not a lot more in control of that the MMU does. I am also willing to bet that they still have things like max extruders hard-coded, which could make things tricky.

Once I get the new MMU (probably September) and have a chance to play with it, I will probably decide what to do with this project, and whether or not it can be adapted to the Mk4. Most likely, given that Prusa has added interaction with the printer for troubleshooting, my screen concept will be unnecessary. This would mean the shortest path to compatibility would be minor modifications to the MMU code (rotation directions, distance between filaments, etc), but probably more significant changes to the actual printer code. Also, given that they moved away from Arduino IDE as a compiler, which I am very familiar with, there will be an additional learning curve for me.

greenfox1505, 2023-08-07:
Would it make sense to re-implement an MMUx12 from scratch? Sure, a MMU3 upgrade would be ideal, but I don't know if it's necessary. Arduinos and stepper drivers are easy to get. I've got a hand full of old RAMPS boards laying around.

grzlus, 2023-10-30:
I'll be upgrading to 3.5, so I'll need MK3 upgrade. I'll reuse as many files from current MMUx12, so the original firmware will be only patched to use correct rotations as we have them in the files. I'll probably abandon that firmware, as it will too much to patch changes made by Prusa and it's easier to start from scratch with firmware.

Dutchy1969, 2023-11-10:
I can’t wait to have a solution for the 4. I appreciate you guys help. I’m not smart enough for this stuff but if there is anything I can do to help you guys out to get this to work on a Mk4 you let me know.

grzlus, 2024-02-23:
@Dutchy1969 I'm waiting for my order to be delivered. I'm getting MK3.5 and MMU2S to  MMU3 upgrade.

In a month I should have it delivered and put together :)

grzlus, 2024-03-06:
@Dutchy1969 good news, my 3.5 and MMU3 upgrade was shipped :)

sithlord48, 2024-05-01:
TLDR: For an MK3.5+ solution or one allowing the user to use the newer MK3S firmware 3.13+ 
This firmware must be rebased upon the MMU's 3.x series, or otherwise implement the bi directional communication so the printer can talk to the mmu in the way it expects. Devs could upgrade their MK3S to firmware 3.13 to test an Mk4 or MK3.5 is not required. 

There are Two main changes for MMU3  
  
1. [[image]](https://www.prusa3d.com/en/product/mmu-pd-board-addon) A board that goes between the mmu and the printer
    - Stabilizes the power to the mmu
    - Replaces the wires that connect the mmu with a single cable. (Store: [Mk3S Wire](https://www.prusa3d.com/en/product/mmu3-printer-cable-einsy/). Currently no Mk4 link) 

2. Firmware changes to both the MMU (3.x series) and the Mk3S(3.13.x)
   - Printers with firmware 3.13.0 or higher REQUIRE the MMU to use 3.x firmware
   - Enables bi directional communication so the printer can now show mmu errors on the screen. and tell the mmu to retry. No more mmu light based error codes to for those mmu issues!!
   - The MMU can be enabled and disabled from the printers menu
   - The printer can send firmware to the MMU to upgrade its firmware when you upgrade the printers (an additional .hex for the mmu firmware is included in the mk3S+ mmu package now)

The rest of the new MMU3 stuff does not matter for this project since its all printed parts. A better idler, housing lid etc.. that will not be used in this project.

For my Mk3S+ MMU2S I got the new power board and I was then able to use the 3.x firmware. Since then i have been using the " MK3S+ MMU3" profile without issue

For the MK4 / 3.5+ the only change to the MMU3 (kit) is the wire you use to hook it up to the printer.

Johnday29, 2024-07-05:
Just checking in here.  Let me say your mods to the MMU2 have been a game changers for my little 3D printing business, so Big thank you to @cjbaar .  That being said I have upgraded a couple of my printers to MK4 with MMU3 and must say it is night and day difference the speed and quality of prints I get, This means I really want to upgrade my MMU2 x 12 to MMU3 x 12.  Has there been any progress in this area?  Would be happy to contribute to make it happen. Thanks!

cjbaar, 2024-09-27:
Sorry @Johnday29. I'm glad you have used the 12x successfully. I have not had any chance to look into what it will take to get this working with MMU3.

Johnday29, 2024-09-28:
Thanks for the reply. Please let me know if I can contribute to the effort
to get this going with the mmu3 software. It runs so much smoother and with
Mk4 compatibility it is so much faster would love a boost to the 12x 😎.

3d-gussner, 2024-09-30:
@cjbaar The new MMU firmware has way less hard coded values and lot of that can be defined in https://github.com/prusa3d/Prusa-Firmware-MMU/blob/main/src/config/config.h and https://github.com/prusa3d/Prusa-Firmware/tree/MK3/Firmware/mmu2/variants and https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/MK3S.h#L673

Also have a look at https://prusa3d.github.io/Prusa-Firmware-Doc/Prusa-Firmware-MMU/index.html

But i doubt that you get the LCD display working as it consumes too many resources.

Here a build output of MMU3 _3.0.3

```
[build] AVR Memory Usage
[build] ----------------
[build] Device: atmega32u4
[build] 
[build] Program:   28370 bytes (86.6% Full)
[build] (.text + .data + .bootloader)
[build] 
[build] Data:       1669 bytes (65.2% Full)
[build] (.data + .bss + .noinit)
[build] 
[build] 
[build] 
[build]    text	   data	    bss	    dec	    hex	filename
[build]   27678	    692	    977	  29347	   72a3
```

cjbaar, 2024-09-30:
@3d-gussner That does look halfway promising. I do see variables for the selector start position and distances. It looks like the idler creates a hard-coded array of positions, instead of just constant motor steps. This can be created, but might take  some tweaking. One thing I don't see is an easy way to reverse the motor directions, since some of mine are on opposite sides than the original. Maybe if you see that, let me know.

The bigger issue, then, is on the printer side itself. Will the printer accept 12 tools? The older firmware was hard-coded to throw away any g-code above T4, and also would only display sinlge-digit numbers on the screen. Any idea what that code looks like for the Mk4 board?

cjbaar, 2024-09-30:
@3d-gussner OK, according to axis.h, `dirOn` might actually reverse the motors

cjbaar, 2024-09-30:
And, of course, there is this warning. :)
```
/// Max number of extruders/tools/slots
/// Beware - if you change this, the EEPROM structure will become invalid and no migration procedures have been implemented.
/// So if you really have to change this, erase your EEPROM content then.
static constexpr const uint8_t toolCount = 5U;
```

cjbaar, 2024-09-30:
See its a quick scan of the buddy firmware showing things like this that makes me think the printer won't cooperate without a lot of changes.
```
enum class Tool {
    Tool1 = 0,
    Tool2 = 1,
    Tool3 = 2,
    Tool4 = 3,
    Tool5 = 4,
    None_Any = PrusaToolChanger::MARLIN_NO_TOOL_PICKED, // Park value
};
```

Johnday29, 2024-09-30:
While this is really mostly above my head at this point I am excited you
all are looking into it. Please let me know if I can assist with anything.
Thanks.

cjbaar, 2024-10-01:
So.... this is interesting.
"To flash your own (or a community-made) unsigned firmware, you’ll have to break the appendix on the logic board (called the Buddy board) and move the jumper in a correct position. Breaking the appendix on the Buddy board is irreversible and voids the warranty of your printer’s electronics."

Johnday29, 2024-10-01:
Well that is Just wrong. Prusa is usually better than that.

cjbaar, 2024-10-03:
OK, I've got some good news, some great news, and some possibly discouraging news.

1) I have a working prototype of the MMU3 (x12) on the Mk4. It requires re-compiling the code with custom values in `config.h`, but no other code changes. It will work as a standard MMU3 with all the new interactive features, meaning the printer will display error codes. The mini OLED screen is no longer active. I prefer my use of the screen and buttons, but putting that into the new code would not be beneficial at this point.

2) The Mk4 firmware, as-is (6.1.3) forwards ALL T-codes from the sliced file. This means the 12x works without any modification to the printer or its firmware (other than Nextruder sensor updates required for the MMU3).

3) I cannot get this to work with my older MMU control board, even though it seems identical to the new one. I think this is probably more bootloader or eeprom-related, but I cannot solve it right now. This means that you would probably need a brand new MMU3 board... whatever that difference is. Your results may vary.

I will post more once I have completed further testing. Of course, all of this also depends on Prusa not making any major modifications that block any of it... like suddenly cutting out extra T-codes (like the Mk3 did). I also don't know what will happen once the Prusa firmware becomes in charge of uploading the MMU firmware, which is supposed to occur in the near future. We will have to see if the printer firmware at that point will allow custom-loaded MMU configs.

cjbaar, 2024-10-03:
@3d-gussner How are you getting that build output you show above? I am using the python build script, as described in the instructions. I only ever get "SUCCESS" or "FAILURE" as an output, which clearly isn't helpful when its the latter.

3d-gussner, 2024-10-04:
> @3d-gussner How are you getting that build output you show above? I am using the python build script, as described in the instructions. I only ever get "SUCCESS" or "FAILURE" as an output, which clearly isn't helpful when its the latter.

I use VScode and some extensions like cmake, cmake integration.

After installing VScode and at least these two extensions execute `./utils/bootstrap.py` to ensure that you have all `.dependencies`.
You should then see in the bottom line `CMake: [Release]: Ready` `<wrench+driver_icon> [avr-gcc]` . Press on `Build`

Building with cmake will output more information.

The `build.py` scripts intentionally only output `SUCCESS` or `FAILURE` to make it easy for beginners.

Hope that helps

Johnday29, 2024-10-05:
Can't wait, I'm very excited!! Thanks for the hard work.

cjbaar, 2024-10-18:
It is done. Works for now. New STL files uploaded for v3.3

Dutchy1969, 2024-10-18:
Man congratulations this is AWESOME, I currently don't have a mk4/mmu3 setup I switched to the XL but it's way overkill for what I do. I might just sell it off and go back to a mk4 with now a MMu3/12 thank you so much!!!!!

sithlord48, 2024-10-19:
do you have the new adapter w/ power stablizer in it ? 

>1. A board that goes between the mmu and the printer
>
>    Stabilizes the power to the mmu
>    Replaces the wires that connect the mmu with a single cable. (Store: [Mk3S Wire](https://www.prusa3d.com/en/product/mmu3-printer-cable-einsy/). [MK3.5+ Wire](https://www.prusa3d.com/en/product/mmu3-printer-cable-for-mk3-5-3-9-4/))

sithlord48, 2024-10-19:
Is there a way to tell the `Old` and `New` board apart ? 
Do we know how many "revisions" there could be ? Or anyidea what we do to make the old board the new board ?

cjbaar, 2024-10-19:
Yes. My older board says "Rev 0.3" and the new one "Rev 0.4.4". Both are "sheep edition." I don't think there were any major changes in hardware that should affect this, but I am not clear. Also not sure if (or how) the boot loader might affect this. The "older" one is running and moving, but just doesn't answer the printer somehow. I am not sure how to go about debugging that.

cjbaar, 2024-10-19:
To be clear, I am not able to get the unit to communicate with the Mk4... even using the stock (unmodified) 3.0.3+896 firmware.

cjbaar, 2024-10-19:
It is very difficult to debug the communication now. In the old firmware, the USB was connected to the same UART channel as the printer, so you could just send/receive commands like T0 with a terminal. Now, that is no longer the case. I am trying to get some kind of debug enabled, but can't get any options to compile.

3d-gussner, 2024-10-22:
@cjbaar I found my first MMU board which is also rev. 0.3 and had also some issue to flash the [original MMU 3.0.3+896.hex](https://github.com/prusa3d/Prusa-Firmware-MMU/releases/download/v3.0.3/MMU2S_MMU3_FW3.0.3+896.hex) file.
There is a sequence you need to follow when flashing an old MMU board with PrusaSlicer.

I can't tell the exact steps and timing when to do what but here what I did:

1. Connect MMU board to MK3S (didn't work with MK4 for me) and MicroUSB to PC
2. Power on MK3S but disable MMU in Settings LCD menu
  a. Need the power from the printer to be able to flash the 32u4 chip. Probably the issue with MK4 and old firmware that the power is disabled to the MMU. On MK3S there is no circuit to cut the power to the MMU while it is possible on the MK4
3. Open PrusaSlicer 2.8.1 -> Configuration -> Flash Printer Firmware -> Select Firmware image
4. Press the reset button on MMU to get it into bootloader mode
5. Press Flash in PrusaSlicer.

Step 4 and 5 are the tricky one, I can't tell exactly WHEN and with which DELAY you have to press the reset and/or Flash.
Don't press the reset button twice to prevent the need of flashing the 32u4 via an ISP programmer.

After few attempts the board is operational.

You can use
- `lsusb -d 2c99:0003 -v` to get the current mmu USB information when in bootloader
- `lsusb -d 2c99:0004 -v` to get the current mmu USB information after bootloader and in firmware

3d-gussner, 2024-10-22:
@cjbaar You have a fork of the MM-control-01 MMU 1.x firmware repository and here all your changes are visible.
You fork of Prusa-Firmware (8-bit printer) doesn't have any changes.
Can you please share your MK2.5/S MK3/S changes in a branch of your fork?
Please also create forks from the MMU 3.x firmware repository https://github.com/prusa3d/Prusa-Firmware-MMU and your changes you made in a branch. I have found https://github.com/cjbaar/prusa-mmu-12x/blob/master/firmware/config-mmu3-mk4-12x.h but it would be way easier for others to see the changes comparing the branches.
Also a fork of https://github.com/prusa3d/Prusa-Firmware-Buddy and a branch with your changes to the MK4 code would be nice.

cjbaar, 2024-10-22:
@3d-gussner  If the firmware flashes, it should run. I don't know that there is anything special about the sequence you mentioned that would affect how it operates. So are you saying you successfully flashed the v0.3 board with mmu3-12x firmware, and the printer sees it?

cjbaar, 2024-10-22:
@3d-gussner If you look at the files I moved into the stl/v1.4 folder, the Mk3 fork is here: https://github.com/cjbaar/Prusa-Firmware/tree/MK3_3.11.0. This is the older one for the Mk3. I have not made any forks or changes for the MMU3.

I can make a fork of the MMU firmware, but I didn't want to get into a pattern of having to fork every version they release. Since there are no actual code changes now, I am hoping it will be easier to just use the config file.

On your last point, there is no firmware changes for the buddy/Mk4 code. I am running stock 6.1.3 from Prusa.

cjbaar, 2024-10-22:
@3d-gussner Here is the new fork for MMU firmware: https://github.com/cjbaar/Prusa-Firmware-MMU-12x

3d-gussner, 2024-10-23:
> @3d-gussner If the firmware flashes, it should run. I don't know that there is anything special about the sequence you mentioned that would affect how it operates. So are you saying you successfully flashed the v0.3 board with mmu3-12x firmware, and the printer sees it?

Yes the MM-control board rev.0.3 communicates with the MK3S and or MK4.

cjbaar, 2024-10-23:
@3d-gussner Just to confirm... the rev 0.3 board you are referring to was previously running the 12x firmware?

3d-gussner, 2024-10-24:
@cjbaar No I never used the 12x firmware. I am just trying to help the community. And as you have issues with your rev. 0.3 board I tried to flash the original MMU 3.0.3 firmware to see why you are having issues.

I am tempted to try to build the 12x MMU3 version just to see what is possible. Not sure that a large 12 color print with thousands of changes will make sense as the purge tower and weight will increase by a looooot.

What is interesting is to use the MMU as a filament feed system for single filament prints. Having multiple spools "pre-loaded" and just start a print would make more sense for me to have more than 5 filaments. Also Spool join is another nice function of the MMU.

cjbaar, 2024-10-24:
@3d-gussner Thanks for the confirmation. I know several people who have upgraded MMU2 boards to MMU3, but I am trying to find someone who was previously running the 12x and upgrading. I was not able to do this, but I am not sure if I have something wrong with my board, or the 12x "ruined" the board somehow (which I don't really see how that's possible).

You are certainly right about the waste; I don't do very large prints with it. I have one print I did early on that was 96 hours just to test it out. And, yeah, the waste is like 10x the volume of the print. But normally, I am printing more like 2d art that is only a few layers thick.

I am not touching the printer firmware for the Mk4. I haven't looked at what it would take to get into that code base now. So, unfortunately -- even though the printer will forward G-codes for all tools from a file -- the printer will not have menu options for loading/etc. So I don't think spool join would work on anything about 5 anyway, and you can't do things like pre-load a specific filament for single-color prints.

I have long proposed that the printer could query the MMU for the number of tools at startup, just like it queries for firmware version, etc. Especially with the expanded protocol now being used. But I don't think Prusa has any desire to do that. I submitted my initial changes to the Mk3 firmware in a merge request, just to allow g-code forwarding about 5 tools, but it was completely ignored.

3d-gussner, 2024-10-25:
@cjbaar Do you have a ISP programmer?  It is also possible to use an ARDUINO Uno as an ISP.
With the ISP you should be able to flash the 32u4 via the 6 pins P5 on the MMU board. https://github.com/prusa3d/MM-control-2.0/blob/master/rev.03/MM-control.pdf

Maybe someone from the community can try to flash the original or the MMU3_12x firmware over the previous version.

cjbaar, 2024-10-25:
I do, but I don't want to rely on people who take on this project needing those tools or knowledge. Plus, as far as I know, there should be no difference in how the firmware gets on there.

Biertram, 2024-10-27:
Hi cjbaar,
I thank you for this update. It is godsend. 

Currently I am running a MK3S+ with an MMU2 and the 1.4 12xMMU-Body. 
As the MK3 firmwares require exactly V3.0.2 on the MMU2, I applied most of your config.h to that version and for now the communication seems fine. Both through the MK3 menu and the USB-Terminal (Octoprint in this case).

I did have to modify the Tcodes.ccp in the MK3 firmware though, as it had a check to see if the Tcode is between 0 and 4.

There was still some fiddling left for me to get the parameters in the config.h right. So thank you again for this update. It really helped a lot.

cjbaar, 2024-10-27:
@Biertram Glad to hear it. I have not tested it on the Mk3, so glad you had success with it. The Mk3 firmware continuing to block T codes is very annoying, since it is such a simple change.

3d-gussner, 2024-10-28:
@Biertram :+1:  Could you share your changes you have made to the Prusa-Firmware?

@cjbaar I was searching in the Prusa-Firmware repository Pull Requests and frankly it looks like your changes or your branch have been deleted and I can't see the differences any more.

cjbaar, 2024-10-28:
Yeah, I am sure they deleted that PR long ago. It was actually at 3.9.x, IIRC. Here is my fork of 3.11: https://github.com/cjbaar/Prusa-Firmware

Biertram, 2024-10-28:
@3d-gussner Sure.

Let me preface this with the info that I am not a software engineer, I just know enough to get myself into trouble.

I currently run my MK3 on the firmware build MK3S_MK3S+_FW_3.14.1_ENGLISH.
Originally this version is meant to run with MMU2 V3.0.3. I faintly remember, that there was some trouble with getting it to work, so I tried to roll back the MMU2 to V3.0.2, which wasn't accepted by the version check of the MK3S+ firmware. 

So these are the changes I made:
Tcodes.cpp:
L14-16:
inline bool IsInvalidTCode(char *const s, uint8_t i) {
    return ((s[i] < '0' || s[i] > '11') && s[i] != '?' && s[i] != 'x' && s[i] != 'c'); //T-Code set from 4 to 11 for 12 filaments
}

MK3S.h
L673;
#define MMU_FILAMENT_COUNT 12

mmu2_supported_version.h
L8:
static constexpr uint8_t mmuVersionPatch = 2; ///Default = 3

errors_list.h
L295:
static_assert(MSG_DESC_FW_UPDATE_NEEDED[szFWUN - 3] == ('1' + mmuVersionPatch));  ///Default = ('0' + mmuVersionPatch)

In the MMU2 V3.0.2 I applied cjbaar's config, except for the stepsPerUnit on the Pulley configuration (config.h L118):
Here I kept .stepsPerUnit = (200 * 8 / 19.147274), instead of .stepsPerUnit = (200 * 8 / 37.384921), as the former worked quite well for me.

cjbaar, 2024-10-28:
@Biertram Nice job

Biertram, 2024-10-28:
Thank you. 

I don't quite remember what the issue with V3.0.3 was, but maybe going back to V3.0.2 might fix your communication issues as well. 

I just tried flashing my MMU with my V3.0.3 version, but this time it worked... Strange...
I did change the version number to go with the check in my MK3S though, so I didn't have to reflash it as well.

cjbaar, 2024-10-28:
@3d-gussner So it looks my communication issues are the result of a bad resistor on the board. The TX line (R2) should be 22ohm, but is measuring well over 1K. For whatever reason, the Mk3 works find with this, but the Mk4 drops bytes.

3d-gussner, 2024-10-29:
@Biertram Thanks for sharing.

3d-gussner, 2024-10-29:
@cjbaar I looked at my rev 0.3 board and `R2` is labeled as `022` the same as `R1` next to it. This matches also the [schematics](https://github.com/prusa3d/MM-control-2.0/blob/master/rev.03/MM-control.pdf) 
With my cheap multi-meter (set to 2000 Ohm or 200 Ohm) I measure 23 - 26 Ohm on both `R2` and `R1` when disconnected from the printer. Never tried to use this board on the MK4 to see if there is a difference.

Replacing the tiny 0603 `R2` resistor isn't fun.

How do you connect the rev 0.3 board to the MK3S? Using the MMU2S cables or new PD board and matching cables which are part of the MMU2S to MMU3 upgrade?

Not sure but the PD board also has some components on the TX RX lines and these may have "tighter" tolerances what is expected.

3d-gussner, 2024-10-29:
@Biertram memory dumps are really hard to read and it doesn't make sense in this case.

>Here I kept .stepsPerUnit = (200 * 8 / 19.147274), instead of .stepsPerUnit = (200 * 8 / 37.384921), as the former worked quite well for me.

This value should be changed as the pulley diameter is different than the stock one.

Biertram, 2024-10-29:
Oh right... I confused pulley diameter and selector angle when I did the configuration...
 
The original value didn't give me any trouble yet. But I'll change it to 37.384921 to account for the diameter.

3d-gussner, 2024-10-29:
@Biertram The new MMU firmware is more forgiving on how much filament has been pushed/pulled. It relies more on the states of the FINDA and FSensor than the actual length pushed or pulled. You may be lucky in the previous configuration that the FINDA and FSensor triggered in the thresholds.

cjbaar, 2024-10-29:
@3d-gussner  > How do you connect the rev 0.3 board to the MK3S? Using the MMU2S cables or new PD board and matching cables which are part of the MMU2S to MMU3 upgrade?

Excellent point. I am connecting to the Mk3 with the legacy connectors, so it is probably the PD board that has issues with the faulty resistor. It's not a big deal for me, as I currently don't plan to upgrade my Mk3, but wanted to make sure it wasn't an issue for others trying to upgrade. Thanks for the input.

3d-gussner, 2024-10-30:
>How do you connect the rev 0.3 board to the MK3S? Using the MMU2S cables or new PD board and matching cables which are part of the MMU2S to MMU3 upgrade?

I used the MMU2S cables to connect the rev 0.3 board to the MK3S and it works.
As I have the new MK3S to MMU3 PD board cable I use that one as the PD board improves the Power Delivery.

I have been looking for the differences of the rev 0.3 vs 0.4.4 and I found that the F2 fuse is different
- 0.3 has a 750mA `b0075`
- 0.4.4. has a 1.5A `b150 24`

Guess that the bigger fuse improved the usage of the MMU2/S on the 12V MK2.5/S printers 

>Excellent point. I am connecting to the Mk3 with the legacy connectors, so it is probably the PD board that has issues with the faulty resistor. 

I measured the resistance with my multi-meter between the MMU connector TX / RX pins and MK3.5/S, MK3.9/S and MK4/S Connector TX / RX pins on the PD board and get ~330Ohm.

This means

```
MMU <-----------> MK3S no PD
TX  -> 22 Ohm  -> RX
RX  <- 22 Ohm --- TX

MMU <----------> PD board <-> MK3S with PD
TX  -> 22 Ohm -> 330 Ohm ---> RX
RX  <- 22 Ohm <- 330 Ohm ---- TX

```

3d-gussner, 2024-11-06:
Check if that is working for you https://github.com/3d-gussner/Prusa-Firmware/tree/MK3_MMU3_12x
I don't have the MMU12x but like the idea and spend some free time on that.

Prusaprinter44, 2024-11-11:
> Check if that is working for you https://github.com/3d-gussner/Prusa-Firmware/tree/MK3_MMU3_12x I don't have the MMU12x but like the idea and spend some free time on that.

Does this link have the hex file for the firmware on the mk3s and the new mmu12 firmware? I am having a hard time finding the files.

Prusaprinter44, 2024-11-11:
> @3d-gussner Sure.
>
> Let me preface this with the info that I am not a software engineer, I just know enough to get myself into trouble.
>
> I currently run my MK3 on the firmware build [...]

Are you currently running the mk3s with the new mmu12/mmu3 update?

3d-gussner, 2024-11-12:
@Prusaprinter44

> Does this link have the hex file for the firmware on the mk3s and the new mmu12 firmware? I am having a hard time finding the files.

No there is no hex file. That is just the source code and I think it would make sense if @cjbaar looks into that, compiles it and publishes it here. It is his project, just trying to help.

> Are you currently running the mk3s with the new mmu12/mmu3 update?

No I don't run the MMU3_12x at the moment.

[7 more replies]