cjbaar/prusa-mmu-12x #20: print request

yahbluez, 2023-09-01:
Hi

I like to add this filament changer system as a kind of his own to this list of filament changers:

https://www.printables.com/model/567217-4-color-transition-module

Is it possible to print the 4 color test and give me the data to add them to the list?

I would give a backlink to this github or to the printables project if you like to have one.

cjbaar, 2024-10-18:
The data would the same as the Mk3+MMU2 or Mk4+MMU3. The new MMU3 firmware is the same as stock Prusa, with different config for the 12x spacing. But the operation is identical, so it wouldn't affect waste or print time.

zonkie, 2024-10-19:
Wow, thanks for putting all this effort Into this project! Maybe i will try it again with the mk4s and mmu3 :)

closed 2024-10-24 by cjbaar