cjbaar/prusa-mmu-12x #21: PrusaSlicer custom G-code?

minu8702, 2023-10-29:
I have managed to build mmu12x and flash firmware on both MK3s+ and new MMU12x.
I am having a trouble where I am getting "File Incomplete. Continue anyway ?" error on the printer screen.
When I try with Octoprint, it says there is no tool exist, e.g. T11 for extruder 12.
I have expanded the number of extruders under 'Printer Setting'>'General'.
Also added below G-code under 'Custom G-code'>'Start G-code' for extruders 6-12, as I thought it was necessary.
M403 E5 F{"" + ((filament_type[4]=="FLEX") ? 1 : ((filament_type[5]=="PVA") ? 2 : 0))}
.
.
.
M403 E11 F{"" + ((filament_type[4]=="FLEX") ? 1 : ((filament_type[11]=="PVA") ? 2 : 0))}

If I use normal printer & MMU preset for 5 filaments, it works perfectly, but once I expand the number of extruders, I get the above error.
Do I require any other changes? or does the printer profile need to be made ground up?

Any help would be very appreciated.

minu8702, 2023-10-30:
Actually, just had to use older version of PrusaSlicer and just worked.

closed 2023-10-30 by minu8702

cjbaar, 2023-10-31:
The printer firmware has a hard-coded value to search for the "end" of the g-code, which is "M84." Slicer adds a lot of comment stuff after that, and newer version of slicer keeps adding more. In my version of the printer firmware, I increased the size of the loopback from 10,000 characters to 25,000. However, it sounds like maybe this is not enough anymore. If you are compiling the printer firmware, you can try increasing the value of END_FILE_SECTION (in config.h, I think). Otherwise, try removing all the comments from the end of the generated g-code file.

`#define END_FILE_SECTION 25000 //number of bytes from end of file used for checking if file is complete`