cjbaar/prusa-mmu-12x #23: Umbau mmu3

Lanzelott76, 2024-11-05:
Hallo 
Wird die Hartware von der mmu3 dort nur genommen oder geht die von der mmu2s auch dafür .
Wäre es möglich eine Klipper Lösung zu finden für den Umbau auf 12 Farben ...das wäre super .
Gruß

cjbaar, 2024-11-05:
I believe the firmware I have posted for the MMU3 should work with the previous hardware, but I have not tested it. There have been no changes in the size or distances between filaments. Also, I am not familiar with Klipper, but generally the printer firmware just needs to forward all T commands to the MMU without restriction.

Lanzelott76, 2024-11-05:
Wenn das so ist werde ich mal die neuen Teile drucken und den Umbau wagen .
Danke für die schnelle Antwort. 😉

3d-gussner, 2024-11-06:
The internal communication between the printer and MMU changed.
In the 1.0.6 version the printer send
`S0` to get the Major version number

The new version 3.0.x has an additional CRC checksum
`S0*c6` to get the Major version number

@Lanzelott76 If you find any Klipper solution that works with the MM-control board please let us know. Viel Glück bei der Suche.

Lanzelott76, 2024-11-06:
Nein , die original Platine geht nicht auf Klipper zu flashen...es gibt nur eine Lösung mit einem anderen Bord (skr mini e3 v2) wo man die Klipper Firmware drauf bekommt und die mmu mit steuert .
Ich werde mal schauen ob man die jetzige Klipper Version umschreiben kann von 5 auf 12 Stationen .
Gruß

3d-gussner, 2024-11-06:
@Lanzelott76 I continue writing in English if that is okay.
I have not worked on Klipper but if you can establish a second serial communication to the MM-conrtol board from Klipper and "replicate" the protocol it should be doable to get Klipper "talking" to stock MM-control board.
I can imagine that it is hard to get Klipper firmware on the 32u4 chip which is used on the MM-control board. It doesn't have that many resources.