cjbaar/prusa-mmu-12x #27: 12x12x5 drive gear

HonzaJaros, 2025-01-07:
Hi,

just noticed the eBay link was removed due to the price, there was like $500 on that.

I spoke with seller of this:

https://www.ebay.com.au/itm/355753143663?mkcid=16&mkevt=1&mkrid=705-154756-20017-0&ssspo=ik-zp92zqca&sssrc=4429486&ssuid=q6af2-FRRrm&var=&widget_ver=artemis&media=COPY

It’s out of stock, but if you text them, they open the QTY just you need and backorder it for you. 1-3 weeks they said. 

12x costed 53 bucks

ino89, 2025-01-07:
Hello together in Germany you can buy the one here 

https://www.roboter-bausatz.de/p/mk7-mk8-extruder-vorschubrad-fuer-1.75-3mm-filament-40-zaehne-5mm-bohrung?sPartner=8&gad_source=1&gclid=Cj0KCQiAvvO7BhC-ARIsAGFyToU8h0YlKSCjXkqhxf6G8wXpc_gWQ21H6zRnCCmpWJUqAq21_ZCM2b0aAsxvEALw_wcB

momorox, 2025-01-08:
@ino89 Unfortunately, it looks like the height is 11mm, not 12mm.

HonzaJaros, 2025-01-08:
[image]

I asked especially

jasonmatthews41, 2025-01-08:
In my opinion, the 11 mm ones work just fine. I have had zero issues, and I have done dozens of prints already.

HonzaJaros, 2025-01-08:
I have ordered them and the rest if needed hardware too so there is no way back and I’ll make it work.

I just wanted other ppl know they provide them even they are not in stock…

just off topic, is there a discord or fb page to discuss about this?

momorox, 2025-01-09:
Nope. Also, I was actually referring to the link that ino89 posted.

MagnusVesper, 2025-05-23:
Would this work? They look more like the ones used on the stock MMU3. But they're 13mm wide.
https://www.amazon.com/uxcell-Drive-Direct-Extruder-Reprap/dp/B07Y9Z77LV

[image]

jasonmatthews41, 2025-05-23:
Those will not work. The entire piece needs to be splined. Some of them will grab more than one filament at a time.

MagnusVesper, 2025-05-23:
Ah, thanks. Does that mean I could use fully splined drive gears of any width as long as they add up to 84mm?

cjbaar, 2025-05-23:
The reason I went with the other style (I forgot what printer brand uses them now) is that each gear works on two filaments. The 6mm spacing was based on the gears being 12mm wide. You just need to make sure a grub screw isn't lined up with one of the filaments.

MagnusVesper, 2025-05-24:
I asked in a maker group I'm in and got some useful info:
A gear with teeth like that is a "spur gear" and the term for having teeth along the whole width is "hubless"
But these aren't actually gears, because the teeth are pointed. They are "feed wheels" or "feel rollers".
This specific kind with a grub screw and straight teeth seems to be mainly used for 3D printing, so just "extruder wheel" may be the best term for them.
Hopefully these names will make it easier to find the correct kind.

AnatomicFlack, 2025-07-04:
Would this work?  The only issue I could see is they are 11mm diameter (which seems to be most of the available offerings anywhere), but the fact that it's wire cut and that the teeth go all the way to the end of the unit seems like it would be beneficial.

https://www.aliexpress.com/i/2251832348720735.html