cjbaar/prusa-mmu-12x #28: MK4 BSOD - Multiple CFSR Errors

momorox, 2025-01-10:
Whenever I try to print something, a few minutes in it crashes. This even happens when I try to print with only one color. I tried the firmware posted here as well as one I built myself. If I disable the MMU and bypass it, the printer functions normally.
[image]

Printer: MK4S
Firmware: 6.1.4
Buddy Board: 37

MMU3
Firmware Version: 3.0.3
MMU Board Revision: 0.4.4

jasonmatthews41, 2025-01-10:
I might be wrong here, but are you running a MK4 flash on your MK4S? It could be that. I am running a full MK4S flash on my MK4S. I've never had a single problem.

momorox, 2025-01-10:
I don't think that's it. The MK4 and MK4S use the same firmware. You set it to the MK4S through the settings.

jasonmatthews41, 2025-01-10:
It's not exactly the same firmware. I found out the hard way, that the MK4S absolutely must have the silicone sock, or you burn up thermistors, and heaters. I went through two thermistors and a heater cartridge before I figured it out. Also, make sure that with the MMU setup, you have to use the regular brass nozzle, and not the high flow one. The tiniest little things make all the difference in the world with this setup.

momorox, 2025-01-11:
None of these are the issue.

momorox, 2025-01-12:
What revision is your Xbuddy Board? You should be able to see it from the MK4S under "Info"

momorox, 2025-01-12:
Is anyone's xBuddy board revision 3.7?

cjbaar, 2025-01-12:
I'm still running 6.1.3 on my Mk4. Can you try that? Or is 6.1.4 a requirement for the "S" model? The MMU firmware used for the 12x is not altered in any way, it only provides different length parameters via the config.h file. Can you ask Prusa what that stack trace means?

momorox, 2025-01-12:
MK4S doesn't require 6.1.4 and I tried 6.1.3. I'll ask, although I think it stands for Configurable Fault Status Register.

momorox, 2025-01-15:
It turns out my xBuddy board was defective. I replaced it and it now works.

closed 2025-01-15 by momorox

Korri, 2025-01-17:
I had the same issue on my MK4S, and upgrading to the `rc1` fixed it for me too, thanks for updating!