cjbaar/prusa-mmu-12x #31: T40 Drive Gears not pushing filament to the Selector

edirian000, 2025-01-24:
Hello, I've tried everything i can think of to resolving this issue but have been unable to. Hoping someone out there can assist me. When trying to start a print the gears aren't catching the filament and pushing it to the selector... Not sure why when taking a closer look everything looks fine but looks like the gears are just rubbing against the filament and not acutally pushing it forward.. Does anyone have a solution to resolving this by any chance?

ino89, 2025-01-24:
Did you pull the screws a little tighter?

edirian000, 2025-01-25:
I did, seems to be pulling a bit more, but now the finda sensor keeps saying it cant trigger... not sure why it wont see the filament passing it.

jasonmatthews41, 2025-01-25:
Probably a stupid question, but did you lower the sensor far enough for it to read the ball bearing?

edirian000, 2025-01-25:
ball bearing?? i dont believe my mmu3 came with one lol, was it suppose to?

jasonmatthews41, 2025-01-25:
Yes. It comes with a ball bearing that triggers the sensor with magnetism. I'm including a couple of photos. Just underneath the sensor there should be the ball bearing to trigger the sensor.

[image] 

[image]

edirian000, 2025-01-25:
wow.. i had no idea, ill try to find it and let ya know the results, thank you for the assistance friends ^_^

edirian000, 2025-01-25:
So the finda sensor is good now! But its back to the gears again, theyre just not wanting to pull the filament.

jasonmatthews41, 2025-01-25:
I would say try tightening the two spring screws holding down the top. Mine works just great with just a couple of threads. If it doesn't grip enough, I would tighten them a half a turn to a full turn. If the selector cannot home, then I would loosen them about a half a turn. It's a love-hate relationship with having the tensions just right. It's the only annoying part of the whole thing. I find that I have to constantly adjust mine. I was thinking about getting nyloc nuts instead of the regular plate nuts.

edirian000, 2025-02-07:
i have tried tightening the screw, but i think i might be possibly missing something? Here's some pictures of where i think something might be missing. If so what is missing though??

[image]

[image]

[image]

jasonmatthews41, 2025-02-07:
Yeah dude. You are definitely missing something all right. You need a ball bearing there.

closed 2025-02-07 by edirian000