cjbaar/prusa-mmu-12x #32: Crashing on filament 6

VSCustoms, 2025-01-25:
Prusa MK4S. Firmware 6.1.4. It will load filament 6 and then crash. Has happened twice so far. 

[image]

VSCustoms, 2025-01-26:
Fixed on 4S using 6.2.0-RC1. 

[image]

closed 2025-01-26 by VSCustoms

jasonmatthews41, 2025-01-26:
Hell yeah dude! You are the one on YouTube that inspired me into this entire thing in the first place.

VSCustoms, 2025-01-27:
> Hell yeah dude! You are the one on YouTube that inspired me into this entire thing in the first place.

Thanks man. Good to know. Honestly it takes a Prusa to the next level. And when just testing this MK4s has my mind blown. The new design changes on the MMU12 just work with the MMU3 and MK4S upgrades.