cjbaar/prusa-mmu-12x #33: Compatibility of MMU3-12x with custom MK3S firmware

hiroloquy, 2025-05-25:
Has anyone successfully gotten the MMU3-12x to work with custom MK3S firmware?

I'm currently using a Prusa i3 MK3S+ with an MMU2S-12x, and I’m interested in upgrading to an MK3S+ with MMU3-12x.
To that end, I’ve installed the following firmware:

- Printer: [MK3S_MK3S+_MMU12x_FW_3.14.1_MULTILANG.hex](https://github.com/3d-gussner/Prusa-Firmware/releases/download/c3.14.1_MMU12x/MK3S_MK3S+_MMU12x_FW_3.14.1_MULTILANG.hex) (by @3d-gussner)
- MMU: [cjb_fw_MMU2board_12x-nodisplay.hex](https://github.com/cjbaar/prusa-mmu-12x/blob/master/firmware/cjb_fw_MMU2board_12x-nodisplay.hex) and [cjb_fw_mmu3_3.0.3-x12.hex](https://github.com/cjbaar/prusa-mmu-12x/blob/master/firmware/cjb_fw_mmu3_3.0.3-x12.hex) (by @cjbaar)

However, the MK3S+ does not recognize the MMU (`LCD menu > Support > MMU: N/A`).
If anyone has experience with this setup or can offer any tips, I’d greatly appreciate it.

Thanks!

Simon-CR, 2025-05-29:
> [@Simon-CR](https://github.com/Simon-CR) Did you upgrade the firmware on a board that was previously flashed as MMU2? That’s where I have had the issue. And it sounds like [@hiroloquy](https://github.com/hiroloquy) maybe running into the same thing.

sadly no, mine started as mmu3.  there must be some differences between mmu2 and mmu3, @hiroloquy, does the prusa mmu3 firmware allow the mmu module to be detected properly ?

3d-gussner, 2025-05-31:
@Simon-CR Please try to connect your MK3S via USB and open a serial monitor like Putty and see what happens when you enable the MMU in LCD Settings->MMU [ON|OFF]

You should see some output like this:
```
echo:MMU2:<S0 A3*22.
echo:MMU2:>S1*ad.
echo:MMU2:<S1 A0*34.
echo:MMU2:>S2*10.
echo:MMU2:<S2 A3*70.
echo:MMU2:>S3*7b.
echo:MMU2:<S3 A380*d9.
```

`echo:MMU2:<S0 A3*22.` Is the response from the MMU that it's major version number is 3. `A`nswer = 3 plus CRC check
`echo:MMU2:<S1 A0*34.` Minor = 0
`echo:MMU2:<S2 A3*70.` Revision = 3
`echo:MMU2:<S3 A380*d9.` Build number = 380

Also always install the firmware with PrusaSlicer especially the printer firmware.

Which board version do you have?

You can also try to send `M709 X42` to erase the eeprom of the MMU, see [here](https://reprap.org/wiki/G-code#M709:_MMU_power_.26_reset)

hiroloquy, 2025-06-01:
Thank you all for sharing such helpful tips!

I had mistakenly assumed that the MK3S+ would automatically detect the MMU2S/3. However, after setting `MMU: On` via the LCD menu, the printer successfully recognized the MMU. I was then able to use both `Preload to MMU` and `Load to nozzle` functions with all 12 filaments.
At the time, I was using [`MK3S_MK3S+_MMU12x_FW_3.14.1_MULTILANG.hex`](https://github.com/3d-gussner/Prusa-Firmware/releases/download/c3.14.1_MMU12x/MK3S_MK3S+_MMU12x_FW_3.14.1_MULTILANG.hex) and  [`cjb_fw_mmu3_3.0.3-x12.hex`](https://github.com/cjbaar/prusa-mmu-12x/blob/master/firmware/cjb_fw_mmu3_3.0.3-x12.hex).

I also confirmed that the printer is properly recognized by PrusaConnect using a Raspberry Pi 4B.
Next, I’m planning to upgrade the MMU-12x body from [v1.4](https://github.com/cjbaar/prusa-mmu-12x/tree/master/stl/v1.4) to [v3.3](https://github.com/cjbaar/prusa-mmu-12x/tree/master/stl/v3.3), and test if full 12-color prints work as expected.

Thanks again, everyone!

cjbaar, 2025-06-05:
@hiroloquy Glad you got it working. I think the switch to turn on/off the MMU from the menus was added in printer 3.14 on.

hiroloquy, 2025-06-08:
@cjbaar 
Oh nice, thanks for the heads-up! I didn’t know about the MMU on/off switch, so I’m glad I found out here.

Over the past few days, I’ve done a few prints with the MK3S+ and MMU3-12x, and your MMU3 firmware works really well with 3d-gussner’s custom MK3S+ firmware. It handled 3–4 color prints without a hitch—and even full 12-color prints worked perfectly!

[image]

[image]

[image]

Also, even with the custom firmware, I was still able to use Prusa Connect and wirelessly send files from PrusaSlicer, which is super convenient.

[image]

Really happy to have a more capable 12-color MMU setup now. Big thanks to you and @3d-gussner!

By the way, I didn’t use a PD board when upgrading from MMU2S to MMU3. For details, please refer to Chris’s video: [Do You Need An Upgrade? - MMU2 to MMU3 - Chris's Basement - 2023 - YouTube](https://youtu.be/JAWqIH_U66s?t=1136&si=din2Zuzi-mcL1v2n)

minu8702, 2025-09-08:
Thank you @cjbaar, @3d-gussner and @hiroloquy all for mentioning the compatibility of the new version with the MK3S+
I would not have known there was a MMU3X12 compatible firmware available for MK3S+. Have been using MMU2X12 for more than a year always wanted to update the firmware for some of the new features. Now my printer is up-to-date and MMU has been updated to MMU3.
Cannot thank enough for the convenience 12 fialment MMU is providing!!

With the compiling of the custom firmware, is there a way to compile with the thermal model for REVO hotends? I am newbie in this category and just went ahead and changed the hotend. It seemed to work at first but obviously caused thermal anomarly and eventually a thermal error. Currently running with the thermal model protection off but want to see if the firmware can be compiled with the correct thermal model.

3d-gussner, 2025-09-08:
@minu8702 Checkout https://github.com/3d-gussner/Prusa-Firmware/releases/download/c3.14.1_MMU12x/MK3S_MK3S+-E3DREVO_MMU12x_FW_3.14.1_MULTILANG.hex or https://github.com/3d-gussner/Prusa-Firmware/releases/download/c3.14.1_MMU12x/MK3S_MK3S+-E3DREVO_HF_60W_MMU12x_FW_3.14.1_MULTILANG.hex

I don't have the REVO nor the MMU12x so can't promise everything is working fine.

3d-gussner, 2025-09-08:
@minu8702 
>With the compiling of the custom firmware, is there a way to compile with the thermal model for REVO hotends?

Ideally you create a copy of the https://github.com/3d-gussner/Prusa-Firmware/blob/c3.14.1_MMU12x/Firmware/variants/MK3S-E3DREVO.h with a new name `MK3S-E3DREVO_MMU12x.h` and change in the new file `#define MMU_FILAMENT_COUNT 5` to `#define MMU_FILAMENT_COUNT 12` 

Then you can use cmake to cleanup the cache and then build it.

closed 2025-09-22 by cjbaar